cyclotron-bonn/nano_util_board #15: Ideas for v0.2

bedakno, 2022-09-07:
- connect to Arduino via RJ45 on the board which can also be used as a power source

bedakno, 2022-11-07:
In a short presentation of the board it was commented that it would be helpful being able to hardware-reset the arduino from afar. That could be done via another RJ45 socket on the board which carries the serial signal and has one lane connected to one of the reset pins of the arduino.

leloup314, 2022-11-07:
Reset can always be done by just connecting to the Arduino headers directly. In any case, a custom cable has to be used / made, therefore it is just as good to use the headers directly imo

bedakno, 2022-11-08:
But how would one activate those pins in case the arduino isn't responding?

leloup314, 2022-11-08:
The hardware reset on the Arduino is done by pulling the RESET pin to GND. Therfore, you can just connect the pin headers RESET and GND to a wire each, solder these to a switch / button  which then shorts these pins on switch/push. This solution is still the same with an extra connector for reset

leloup314, 2022-11-08:
Or are we talking about different things ? ;)

leloup314, 2022-11-09:
Ok, I mean the same :) We can just use a cable as I said, patch it through the patch panel, and connect it to the GND and RESET pins, the short it via a button in the control room

leloup314, 2023-01-20:
Closing since v.02 is over ;)

closed 2023-01-20 by leloup314