cyclotron-bonn/nano_util_board #16: BUGs in v0.1

leloup314, 2022-09-17:
- Footprints of resistors appear to be 0402 instead of 0603 
- Voltage divider potentials are switched: we used to measured the voltage drop over the NTC, now we measure over the fixed resistor (easily fixed in firmware)
- Maybe I2C issue (not working so far)

We received 4 PCBs, the NTC readout is wroking fine, the I2C to the R/O board is not yet working, the frequency counting needs some testing ;)

leloup314, 2022-09-17:
Also: when ordering the PCB, the company said that the _drill information_ was missing in the Gerber files (which were then added by our electronicslab)

bedakno, 2022-09-18:
Alright, I'll work on it. Are there any other enhancements I should add? See issue #15.

bedakno, 2022-09-18:
> Also: when ordering the PCB, the company said that the _drill information_ was missing in the Gerber files (which were then added by our electronicslab)

Oh, sorry for that. There is indeed an option for that in the export menu. So that should be a none issue next time

bedakno, 2022-09-21:
Should I swap NTC and 10k so that it'll be as intended before?

leloup314, 2022-09-21:
Yes please

bedakno, 2022-09-21:
Hopefully I fixed those issues. For now, I did **not** change the connection to the I2C RJ45, as I don't know how the correct pinout is. But changing this is no biggy as just the lanes would need to be changed.

leloup314, 2022-09-22:
I have the feeling that the I2C bus on the PCB is not working properly. We have a BA student who exactly uses NTC readout and I2C at the same time, using a breadboard. There it works whereas it does not with the PCB. So maybe the pinout for the RJ45 is correct but I2C is generally not working. Would be good if you meet with the BA student to check the breadboard schematic

leloup314, 2022-10-10:
@bedakno can you comment on the status of the PCB? Maybe a summary of what bugs have been identified and fixed and what has been tested?

leloup314, 2022-10-12:
It is good practice to `commit early & commit often`, would also be nice if you could push the fixes ;)

closed 2022-10-14 by bedakno