cyclotron-bonn/nano_util_board #21: DOC: write short documentation for basil

leloup314, 2023-01-20:
The `nano_util_board` is also used in SiLab by some projects. Here, the firm- and software are provided by the [basil DAQ framework](https://github.com/SiLab-Bonn/basil). A short PDF should be written (~ 1-2 pages) that can be hosted here as documentation.

bedakno, 2023-04-20:
I will begin to write documentation. Now that v1.0 has been reached..

bedakno, 2023-09-12:
I finalised the Wiki. Please have a look at it. I added quite a lot of text with the i2c programming (docstrings). I don't know if that's necessary.

bedakno, 2023-09-12:
Additionally, I didn't find the link to the ``.py`` script for the temperature readout on Basil, I only  found the ``.ino`` file.

leloup314, 2023-09-13:
Looks nice! Would be cool if you could add edits of the photos of which jumpers have to be present for what operation mode (e.g. i2c, ntc). Otherwise it is nice, the docstrings can stay. For basil, you can link to the examples

bedakno, 2023-09-18:
I edited the figures so that they include markings for each usecase.