cyclotron-bonn/nano_util_board #22: BUG: Frequency counting pin wrong

leloup314, 2023-03-01:
The pin which can be used fro frequency counting on the nano is **D5** aka pin 8, **not** A5. The frequency counter input should be rerouted to D5 and a new version should be made. The problems can be omitted by connecting D5 to the frequency input via the respective pin in the 2x5 header

leloup314, 2023-03-02:
@bedakno please adjust the schematics accordingly when you find the time ;)

bedakno, 2023-03-02:
I corrected the mistake in the last commit (I forgot to reference the issue in the commit) and updated the schematic and PCB. New gerber files including drill holes have been uploaded. Also a new version was drafted to be released when the changes are found to be sufficient.

leloup314, 2023-03-03:
Nice, thanks! Now we don't need the 2x5 header anymore but a 2x4 header will suffice. Did you change that as well? I guess we can go for v1 if this is fixed.

closed 2023-04-20 by bedakno